MobileRoboticsSkoltech/OpenCamera-Sensors #35: tags & versions do not match

IzzySoft, 2021-07-06:
Looks like the APKs to the latest 3 releases are all identifying with the very same version: 1.48.3, versionCode 80. That means users having a previous version installed won't get the newer ones offered as updates (as `versionCode` is identical). And also means you cannot tell apart what version was used with errors being reported. Would it be possible to fix this? Thanks!

azaat, 2021-07-08:
Thanks for the feedback, we've overlooked that part! So, from now on we will increase version number and code with every release (tag) - as in pr #36.

IzzySoft, 2021-07-08:
Great – thanks a lot! Besides, since March your app is [available in my repo](https://apt.izzysoft.de/fdroid/index/apk/com.opencamera_extended.app) (which is how I noticed the versioning issue). This not only makes it easier to discover your app – but thanks to your fix also to keep it updated on-device, as updates are fetched within 24h of your attaching the APK to a new tag and then announced on the devices of those using my repo and having the app installed. If you wish to link that from e.g. your Readme, you're welcome using [this badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) for that purpose :smiley:

closed 2021-07-09 by azaat

IzzySoft, 2021-07-09:
Thanks for adding the badge! Maybe you want to point the link to [this URL](https://apt.izzysoft.de/fdroid/index/apk/com.opencamera_extended.app) – so when clicked on the badge, it opens the page of your app there instead of the badge itself?

azaat, 2021-07-09:
Yes, that's better! Thank you for the link and for publishing the app

IzzySoft, 2021-07-09:
Yes! Very fine now, thanks for fixing! And gladly done from my side; I'm not an Android dev, so this is my way to contribute to the FOSS community :smile:

PR #36 (2021-07-09): Update versioning in AndroidManifest.xml
Closes #35 

From this version tags will match in releases