MobileRoboticsSkoltech/OpenCamera-Sensors #56: Stop video with IMU recording

HarborC, 2021-11-30:
Thanks for your great application. There is a strange problem. The error message “Stop video with IMU recording”  displays in the screen, while I record video with imu recording. Why this happens?

azaat, 2021-11-30:
Thank you for the feedback! This could mean that there was an error during video recording initialization, which led to immediate stopping. I guess there are some errors in the initialization process which are not (yet) reported in the UI, so it would be great if you could share the app logs of when this happens (for example, run application with [logcat](https://developer.android.com/studio/command-line/logcat))

HarborC, 2021-12-01:
[logcat_12-01-2021_08-54-11.txt](https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors/files/7629841/logcat_12-01-2021_08-54-11.txt)
Thanks, I used Logcat Reader to record  the app logs of when this happens.