MobileRoboticsSkoltech/OpenCamera-Sensors #64: No RecSync settings

l1346792580123, 2022-02-14:
Thanks for sharing this great work! In order to synchronized recording on multiple smartphones, the "Use RecSync" switch needs to be enabled. However, I can not find "RecSync settings" in the preferences page. My perferences page is similar to
[image]. And there are only 4 buttons when I open this app. The synchronization button is missing. Does this releated to the type of phones?

TimPushkin, 2022-02-14:
Do you use the apk in Releases or the latest version from `master` branch? The thing is that RecSync is currently in the `master` but has not yet been released as an apk.

I hope an apk will be released soon, but now you can build one by yourself from the `master` branch. And then your preferences should look like this: 
[image]

The switch you need is located in "RecSync settings..." and should be available even if your device doesn't support RecSync -- then it will be greyed out.

l1346792580123, 2022-02-14:
Thanks for your reply. I have downloaded the apk from https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors/releases/latest/download/app-release.apk. I will try to build the apk by myself.

closed 2022-02-14 by l1346792580123