MobileRoboticsSkoltech/OpenCamera-Sensors #66: failed to synchronized recording

l1346792580123, 2022-02-15:
I have downloaded the latest apk and set "Use RecSync" successfully. However, when I press the settings synchronization button, the leader phone try to boardcast the settings, then the lear camera black and it prompts that the video recording failed and that camera has a severe error. The only sync setting that is boardcasted is exposure time. The type of the lear phone and client phone is different. Does it matter?

TimPushkin, 2022-02-15:
Thanks for the report!

I can reproduce the error on the devices available to me, so it would be of great help if you could share the logs of the app produced when this happens (using [logcat](https://developer.android.com/studio/command-line/logcat) for example).

I assume that some of the settings broadcasted are not supported by the device receiving them, so the app fails to apply them. This is supposed to be detected by the app though and should not lead to errors.

For now you could try to uncheck all the setting to be synchronized (except of exposure time which cannot be unchecked), switch both device to the video recording mode (settings synchronization also does this, so by doing it manually some potential errors could be eliminated) and then try to run the synchronization again.

> The type of the lear phone and client phone is different. Does it matter?

This could lead to worse synchronization in the resulting videos, but should not lead to any errors in the app.

l1346792580123, 2022-02-16:
Thanks for your advice. I have used logcat to record the app log. Here is the log file [log.txt](https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors/files/8076027/log.txt).

TimPushkin, 2022-02-18:
Could you please also provide the debug information about your device from Preferences - About?

l1346792580123, 2022-02-19:
Here is the debug information from Preferences-About [about.txt](https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors/files/8101691/about.txt). Should I export the information when I start Wi-Fi hotspot and connected by other phones? The uploaded file is exported without other phones connected.

TimPushkin, 2022-02-19:
Thanks, this information is enough for now, I'll look into it. The problem seems to be in the interaction with the camera: according to the logs, settings application itself looks fine, but the app fails to prepare the camera for the subsequent recording. Currently I'm not sure what's causing it, but it's strange that OpenCamera detects your devices Cemera API as `Camera API: None`.

l1346792580123, 2022-02-21:
This may not be a problem, here is the new about file [about.txt](https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors/files/8105605/about.txt). The Camera API is Camera2. I don't know why the about file is different.