MobileRoboticsSkoltech/OpenCamera-Sensors #69: question about phase alignment

l1346792580123, 2022-03-30:
I have used the [script](https://github.com/MobileRoboticsSkoltech/RecSync-android#extraction-and-matching-of-the-frames) to align the video. But I have found that only a few frames (less than ten) can match. I changed the THRESHOLD_NS to 10000000 and more frames can match while still some frames could not match. So I enabled the phase alignment to get a more synchronized result. But I found that the result are similar. I found some problems in using phase alignment. When I press the phase alignment button, the results showed on two phones are different. One shows that the sync was successful and the other shows that it failed. And I tried to press phase alignment button some times, phase error did not become green and is greater than 10 ms. How can I align the two phase and get synchronized videos? Is it a problem with the Wi-Fi hotspot or different type phones?

TimPushkin, 2022-03-30:
The synchronization status on a device displays the status of the device itself, so any phone can be synchronized or not independently from the others.

Does the phone having problems support real-time timestamping (see [here](https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors#synchronized-recording-on-multiple-smartphones-recsync))?

We actually had similar problems when a phone caoldn't synchronize its phase even though it meets all the requirements, but for now we were not able to identify the source of the problem. It might be helpful, if you provide your phone's model, but I'm not sure that it is something that can be fixed right now.

l1346792580123, 2022-03-31:
Thanks for your reply. I don't quite understand what is the synchronization status of device itself. Synchroniztion status in my understanding is similar to TCP which is a bidirectional status. Anyway, do you mean when all devices show synchronization succeed is real synchronization succeed?

I have checked that and the timestamp sources of all the devices are realtime.

The phone models I use are HuaWei mate 20 and Honor Magic 2. I look forward to your updates and thanks again for sharing this excellent work!

TimPushkin, 2022-03-31:
Currently each device synchronizes itself to a preconfigured phase, so, yes, devices are truly synchronized when each of them succeeded.

We didn't test our work on Huawei and Honor, but the on devices we did test it, there was a correlation between success rate and exposure value set on leader (clients then receive and use it too). You can try to manually set it as low as possible (on the devices we did the tests we usually set the exposure to 1/32 and less), compensating by setting greater ISO values.

l1346792580123, 2022-03-31:
Thanks for your advice. I manually set ISO and exposure value on leader phone and broadcast to client phone. The ISO and exposure value are same when broadcasting the setting, however, when I press the phase alignment button the ISO and exposure value start to fluctuate and still fail to align. Is there something I am missing?

TimPushkin, 2022-03-31:
Hmm, no, this is some strange behavior. The settings that are synchronized among leader and clients (exposure is always one of them, for instance) should be fixed after settings synchronization. Fluctuations of exposure might be the cause of the failure to align phases. Does it happen on a client or on a leader?

l1346792580123, 2022-03-31:
> Hmm, no, this is some strange behavior. The settings that are synchronized among leader and clients (exposure is always one of them, for instance) should be fixed after settings synchronization. [...]

Both leader and client phones. The ISO and exposure value are fixed after broadcasting, and start to fluctuate after pressing phase alignment button.