MobileRoboticsSkoltech/OpenCamera-Sensors #77: Get timestamps for each frame of video

yhabib29, 2022-07-27:
Hello,

I am trying to record a video with synchronized IMU. I can get the IMU data with all timestamps but the application does not provide timestamps for each video frame.
Is it currently possible with the application ?

Thanks in advance !

anastasiia-kornilova, 2022-07-27:
@yhabib29 , could you, please, share logs produced by [logcat](https://developer.android.com/studio/command-line/logcat) when run application? Is this behavior reproducible on other smartphones?

yhabib29, 2022-07-28:
@anastasiia-kornilova is the application supposed to save a file with timestamps for each frame or to save all frames instead of a video ?
I can share logs produced by logcat but do you have any filter to apply because the output of logcat contains all the system background logs ...

anastasiia-kornilova, 2022-07-29:
@yhabib29 , in default behavior it should be recorded video file and file with timestamps, additionally, you can enable "save frames" option as described here -- https://github.com/MobileRoboticsSkoltech/OpenCamera-Sensors#usage. 
Yes, full log will be enough for me.

yhabib29, 2022-08-04:
I see, but what I get is:
- Recorded video file.
- some images ("save frames" option).
- Accelerometer data (CSV file).
- Gyroscope data (CSV file).
- Magnetometer data (CSV file)
- IMU timestamps (CSV file).

None of these files contains the timestamp of each frame. It would be really convenient to have an option to save all frames with timestamps or a CSV file specifying the timestamp of each frame ...

anastasiia-kornilova, 2022-08-04:
@yhabib29 , without logs it is difficult to understand what the source of the problem. About saved images -- they should be saved with filename that is a timestamp. Saving all images is too heavy for processor, because in that case optimized codec is not used.

yhabib29, 2022-08-04:
Ok I understand, it is indeed heavy to save all all images.
Then, what about saving the timestamp of each frame in a CSV file like the [VIRec](https://github.com/A3DV/VIRec) ?

I will try to record the log but I do not think it is relevant if your app is not supposed to save frame timestamps, is it ?

anastasiia-kornilova, 2022-08-04:
@yhabib29 , csv file with timestamps should be available in the same folder along with sensor records, app is supposed to save them and we have it working on couple smartphones. So this issue occurs currently on your smartphone and should be debugged.